amichai-bd/riscv-multi-core-lotr #16: GPC_4T - Enable Alive Test - Basic ALU Operations

amichai-bd, 2021-04-13:
Enable Alive Test - Basic ALU Operations
- [ ] Add/Sub/Shift Imm/Register
- [ ] Jump/branch
- [ ] Write results to MMIO
- [ ] Test Self-Check -> if the MMIO DATA is as Expected Write to MMIO_Succesful Bit.

adlv888, 2021-05-11:
great news!! LW,SW,ADD,ADDI,JAL tested successfully! few bugs fixed in core related to core
d_mem artificially added the size of stack pointer 288 in order to access addresses. updates on Wednesday

amichai-bd, 2021-05-11:
[image]
i see what you did - but I don understand.
you added an offset for each write - and you added an offset for each read.
i don't understand what affect they have - just an offset in memory.

adlv888, 2021-05-11:
I will explain tomorrow. in brief : the address input had the value "-266". it is illegal index for "next mem". without it its not working.
its temporary only to check . we will discuss tomorrow

amichai-bd, 2021-05-13:
@SaarKadosh 
Please push to origin the Alive test you work on.
- alive.c   
- alive.c.s    
- alive.elf   
- alive_lef.txt   
And you may remove the old test

SaarKadosh, 2021-09-29:
Was done and not closed.

closed 2021-09-29 by SaarKadosh