amichai-bd/riscv-multi-core-lotr #22: GPC_4T - break down the Instruction at Q102 and not Q101 - will save  many FF

amichai-bd, 2021-04-25:
Looking at the diagram (and the code) we can see that Many of the samples at 101->102 are just different bits of the Instruction.  
[image]  

We should sample the "raw" Instruction 101->102 and at 102 we will break down the bits into the relevant structures:   
- Immediate B/J/U/I/S.  
- RegWrPtr
- Funct7
- Funct3  

This will save many redundant FF - and as far as I can tell won't affect the Tpd timing in Cycle 102.  
Let me know what you guys think.

adlv888, 2021-04-27:
with this implementation the decode stage will spread on 2 pipe stages 101 102 and can be little confusing. I personally prefer order but I will consult with Saar.

adlv888, 2021-05-05:
accept both ABD and SK opinion. Done

closed 2021-05-05 by adlv888