amichai-bd/riscv-multi-core-lotr #39: GPC_4T - Add Interface of gpc_4t.sv

amichai-bd, 2021-05-13:
Please see the ring controller interface with the gpc_4 there:  
https://github.com/amichai-bd/riscv-multi-core-lotr/wiki/HAS-RING_CTRL-RC  

Please add the interface to the gpc_4t.sv as defined by RC HAS.  

In the test bench gpc instance:  
- input - strap to '0;  
- output - leave dangling.

### Within the gpc_4t.sv - at this point no need to connect at all. 
Eventually will be connected to :  
1) D_MEM + I_MEM. (F2C ->Fabric to core)  
2) Load/Store interface from core - in case the LW/SW request was not for the local memory (C2F -> Core to Fabric)

amichai-bd, 2021-05-13:
Please note - this task is important so others can start integrating the GPC_4T into the LOTR_TILE -> and to the Fabric LOTR.
@adlv888

adlv888, 2021-05-17:
done. hope its fine

amichai-bd, 2021-05-17:
I will try to integrate the GPC and the RC and I will let you know.

closed 2021-05-24 by adlv888