amichai-bd/riscv-multi-core-lotr #47: GPC_4T - move the AliveTest to the correct dir and please also commit the .s, .elf elf_.txt

amichai-bd, 2021-05-17:
move the AliveTest to the correct dir and please also commit the .s, .elf elf_.txt
see your commit 369087db3ab720637d71dbe7c56711c2d2feab6f

DL8, 2021-05-17:
If I understand the scripts correctly, those files are generated based on the C source file. Is that correct? And if that's indeed the case, why are they in the repository in the first place? As a rule of thumb, it's a bad idea to keep auto-generated files/build outcomes in the repository, because you'll easily find yourself with inconsistent copies of the same thing.

The best approach is to keep only the source files and build scripts in Git, so only one version is maintained

amichai-bd, 2021-05-17:
You are correct - the ".s", ".elf", "elf_.txt" ".sv" are all generated based on the C source file.
In most cases, I agree with the rule of thumb "bad idea to keep auto-generated files/build outcomes in the repository" - 

But In this case, I want us to "commit" the generated files.
This is due to some contributors (mainly from the HW part) don't have the RISCV toolchain.
Committing the files the will allow anyone with a SystemVerilog executing/simulation env to test and run the programs we already compiled, Link, and turned into SystemVerilog memory format.

Whenever anyone makes changes in the "C" code they must fun the compile and generate scripts - and commit all changes.

SaarKadosh, 2021-05-27:
Files were moved to the alive dir (under apps). for some reason alive/ dir name won't change to tests/

closed 2021-05-27 by SaarKadosh