amichai-bd/riscv-multi-core-lotr #51: Establish validation infrastructure for software stack

DL8, 2021-05-24:
An emulated environment should be created to enable software execution without hardware dependency.

One option (if support is available) is to use Unicorn engine and implement hooks for memory and CR access. Otherwise another approach/RISC-V emulator should be considered.

DL8, 2021-06-10:
Unfortunately, Unicorn engine doesn't support RISC-V, so need to find another utility

amichai-bd, 2021-06-22:
I think using our Design to debug the SW is possible and usfull.
The Vlog/Vsim that compile the HW and simulation of the program running on our core can create any log/indication we need to debug HW and SW.

Once we are confident with the correctness of our HW design (by RTL0.8) we will try and start "debugging" SW with our trks and logs.

closed 2022-06-08 by amichai-bd