amichai-bd/riscv-multi-core-lotr #61: GPC_4T - Heap enabling

adlv888, 2021-06-01:
how to generate the Systemverilog heap data so we can "load" it with the test

DL8, 2021-06-09:
I think the linker script should create a data segment, such that the data section will be in D_MEM. I don't know how to distinguish between global and local data with this approach though.

Yesterday in the library I added compilation-defined memory layout and the next step is to make it create a linker script that initializes memory according to the memory layout (e.g. place global variables in global data and in the future local data in local data sections)

adlv888, 2021-08-17:
after discussion , 2 issues regarding this:
1. the "HEAP" is  defined in this address:
[image]
all 4 threads takes data from this address which belong to certain thread and copy to seperate registers which is little wierd:
[image]
in this there is a 10 sized array.

2.bigger issue is with bigger arrays. in a 20 sized array the assembly code uses "memcpy" function which doesn't work and d_mem is empty besides the heap:
[image]

adlv888, 2021-08-17:
correction for 2 - it works and write to memory but after few cycles it overwrite the data with "xxxxxxxx"

adlv888, 2021-08-17:
we want the "HEAP" to be on shared mem?

amichai-bd, 2021-08-17:
I think we want this region of memory to be loaded to a shared dedicated area - should not be reliteive to the Stack Pointer offset.
All threads can access this area and modify this data independently - In SW you need to make sure they use this data in a coherent way - so they won't override each others modifications.

amichai-bd, 2021-08-17:
I agree we might have a problem here - if each thread will allocate parts of this area independently they might override each other.
We should talk about what's going on in the data_mem allocation when you set arrays within the "Main" vs in the specific thread.

adlv888, 2021-08-24:
[image]
Heap worked with a lot of data. closing issue

closed 2021-08-24 by adlv888