amichai-bd/riscv-multi-core-lotr #79: GPC_4T - script to run multiple tests

adlv888, 2021-06-15:
create python or bash script to run power shell cmds

DL8, 2021-06-15:
Why would you need Python/Bash to run PowerShell stuff? If you want to use PowerShell, why don't you write PowerShell scripts directly? Otherwise you can just write automation in Python/Bash, no need to add another layer of complexity

amichai-bd, 2021-06-15:
we want to run the vlog & vsim from the POWERSHELL.
But we want some kind of wrapper so we can run multiple system Verilog tests with deffrenet I_MEM loaded - different tests.

I think we should wrap the vlog/vsim commands with python that will allow us to have a simple UI with the systemverilog tests.

Using PowerShell may work - but i think having a Python wrapper is very acceptable and common.

DL8, 2021-06-15:
When you say PowerShell, do you mean those few commands from the README in modelsim? They don't consist of anything specific to PowerShell, so this term is very confusing (you can execute it from PowerShell, but if do it from your favorite shell).

And yes, it's very common (and I would even say mandatory) to write scripts to automate stuff. Python is one useful tool for it

adlv888, 2021-06-22:
i eventually created a powershell script. came out very nice

amichai-bd, 2021-06-22:
@adlv888 - sounds great!
Excited to see it in use!

closed 2021-06-22 by adlv888