amichai-bd/riscv-multi-core-lotr #85: Thread synchronization at reset is required for proper initialization

DL8, 2021-06-17:
Currently the library can't initialize memory layout, because all threads start up and there is no way to synchronize them: only one thread should perform initialization, but since memory is not initialized, it's not possible to establish a communication channel between threads (it should be done by a single thread). To address this, several approaches can be taken:
1. Hardware approach: change reset flow such that only one thread will start running. Software will be responsible to initialize what's needed and start other threads. This may be more complicated from hardware perspective, but will be easier to work with
2. Software approach: currently apps using the library are built with a text section (code) and data section (including GPC params, which will be loaded into global data in D_MEM). This will solve the problem on first launch, but will break on reset

### Discussed in https://github.com/amichai-bd/riscv-multi-core-lotr/discussions/52

amichai-bd, 2021-06-22:
We will add a couple of registers in the CR Space (which are deterministic at Reset) that will be used for general use for SW.
GPF (general purpose flag) - single bit indication
GPSP (general purpose scratchpad)

DL8, 2021-06-22:
Simplifying the definition for starter: 4 registers, each is 32 bits and reset value is zero. For naming, let's call them scratchpads (`SCRATCHPAD<i>` for `i=0,...,3`). I'll add that to the wiki.

As for other definitions (e.g. non-zero reset value, GPF), they may be useful in the future, but at the moment I don't see any use case for them. Also, I feel adding them makes the interface more complicated, because software will have to match hardware-enforced structure and hardware will need to define behavior in case of setting invalid values

DL8, 2021-06-23:
Library is updated to use scratchpad registers as documented in the wiki

closed 2021-06-23 by DL8