amichai-bd/riscv-multi-core-lotr #93: Pad code segment with EBREAKs

DL8, 2021-06-21:
Currently libgpc linker creates the .text to be just as big as the compiled binary requires. When loaded into I_MEM, the rest of it will end up as garbage. To improve application debugability, the .text segment should be padded with EBREAKs, so execution will stop if PC goes out of bounds.

Open: is it applicable to non-debug builds as well?

amichai-bd, 2021-06-22:
This is very good practice for I_MEM - let's figure out how to integrate this to the linker/crt0

DL8, 2021-06-22:
I got into strange troubles when trying to implement this padding in linkage. However, I found that zero is also good for this purpose: it has the structure of a compressed instruction (C extension), but it doesn't map to any real instruction (in disassembly it's shown as "unimp", which is not a real instruction). Bits `[1:0]` are used to indicate whether the instruction is compressed or not, so it won't be possible to get a valid opcode even if C extension is not supported. Therefore, this is good for this purpose (and much easier to implement).

See also:
https://github.com/riscv/riscv-asm-manual/pull/20