amichai-bd/riscv-multi-core-lotr #95: GPC_4T - execute 4 threaded program

adlv888, 2021-06-22:
compile & simulate a  program that simultaneously execute  4 different calculations

[image]

adlv888, 2021-06-22:
[image]
so cool

amichai-bd, 2021-06-22:
Good job - This is great!
It would be even cooler to make them work together to calculate together to accelerate calculation time.
Like a distributed Computation of a prime factor of a number. (each thread will search for prime factors within a Range(

DL8, 2021-06-22:
Could the Sieve of Eratosthenes can be an interesting example for that? In its basic approach I feel it can be executed concurrently, but the naive approach may lead to some redundant work

https://en.wikipedia.org/wiki/Sieve_of_Eratosthenes

closed 2021-06-26 by adlv888