amichai-bd/riscv-multi-core-lotr #165: LOTR - Add Tracker file for all Ring transactions.

amichai-bd, 2022-01-08:
Add Tracker file for all Ring transactions - catch it in each Tile output.
For both Ring Req & Rsp Channels.
The tracker should be a single file for all ring transactions.
fields:
1. timestamp
2. Tile ID
3. Channel
4. requestor
5. Opcode
6. Address
7. Data

adlv888, 2022-04-07:
managed to add tracker to all transactions for specific tile. maybe i didnt get your idea regarding "ring interface and not tile"

amichai-bd, 2022-04-07:
"ring interface and not tile"
Meaning all 8 RC interfaces:
    input   RingReqIn*Q500H    ,
    input   RingRspIn*Q500H    ,
    output  RingReqOut*Q502H     ,
    output  RingRspOut*Q502H     ,
    input   F2C_Rsp*Q500H      ,
    output  F2C_Req*Q502H      ,
    input   C2F_Req*Q500H      ,
    output  C2F_Rsp*Q502H      ,