amichai-bd/riscv-multi-core-lotr #166: LOTR - Add Tracker file for a single RC

amichai-bd, 2022-01-08:
To ease debugging, Please make a tracker for each RC:
single file for each RC.
should track all 8 Interfaces!! (Channel, Opcode, Address, Requestor, Data)
1. RingReqIn
2. RingRspIn
3. RingReqOut
4. RingRspOut
5. C2F_Req
6. C2F_Rsp
7. F2C_Req
8. F2C_Rsp

adlv888, 2022-04-07:
ok so this one i did

amichai-bd, 2022-05-01:
@adlv888 - reminder, Please send a screenshot here & close this issue.

adlv888, 2022-05-01:
[image]

adlv888, 2022-05-01:
not closing yet, adding some fixes we talked about (like ---- on read "requested data")

adlv888, 2022-05-21:
[image]

closed 2022-05-21 by adlv888