amichai-bd/riscv-multi-core-lotr #172: LOTR - RC debug

adlv888, 2022-04-07:
[image]
seems that RC is having some issues with transactions
need to debug

adlv888, 2022-05-01:
found the issue with the READ. ridiculous and amazing issue :
[image]

"d_mem" had no signal called "pre_q_b". only the assign with no declaration. I added this declaration and it fixed(Line 39):
[image]

 how is this possible ? how it was compiled until now ? WTF?!

amichai-bd, 2022-05-01:
We should start running more advanced "Static checks" that make sure we dont have this kind of issue again.
A good option is to use the "quartos compiler" - this will notify us in case we have any other similar violations.

closed 2022-05-01 by amichai-bd