amichai-bd/riscv-multi-core-lotr #201: LOTR - make a 4 thread test with VGA feedback

amichai-bd, 2022-07-09:
To avoid problem in the ring. Make a test with a single core - 4 threads.

Split the Display to 4 squares. Each will be written from different thread.

A rectangles are generated by writing 0xFFFFFFFF
1) counter - represented by number of rectangles.
2) rotating rectangle around the perimeter
3) simple patterns switching (checkers, lines etc.
4) spell hello using rectangles

amichai-bd, 2022-07-28:
Please program this:
https://www.youtube.com/watch?v=ZZuD6iUe3Pc

adlv888, 2022-08-06:
[image]

amichai-bd, 2022-08-18:
The example you made looks great!
[image]

But I think it's still missing a header per thread:
- "Thread 0 - Bubble Sort"
- "Thread 1 - Quick Sort"
- etc.

closed 2022-10-18 by adlv888