amichai-bd/riscv-multi-core-lotr #270: UART - add option to load sequence of instruction to the python

amichai-bd, 2023-01-19:
Here are a couple of examples of files sequences:
core1_freeze.txt:
```
W  ,  0x01C00150  ,  0x0
W  ,  0x01C00154  ,  0x0
W  ,  0x01C00158  ,  0x0
W  ,  0x01C0015C  ,  0x0
```
core1_realease.txt:
```
W  ,  0x01C00150  ,  0x1
W  ,  0x01C00154  ,  0x1
W  ,  0x01C00158  ,  0x1
W  ,  0x01C0015C  ,  0x1
```

load_fpga_main_program.txt:
```
//===CORE 1&2 freeze PC0,PC1,PC2,PC3====
`include core1_freeze.txt  // if we can support this, it will improve things
`include core2_freeze.txt  // we can explicitly write the freeze instead if `include

//===CORE 1 load instruction & data====
WB ,  0x01000000  ,  ../../../../verif/Tests/fpga_main/fpga_main_inst_mem_rv32i.sv
WB ,  0x01400000  ,  ../../../../verif/Tests/fpga_main/fpga_main_data_mem_rv32i.sv

//===CORE 1 load instruction & data====
WB ,  0x02000000  ,  ../../../../verif/Tests/fpga_main/fpga_main_inst_mem_rv32i.sv
WB ,  0x02400000  ,  ../../../../verif/Tests/fpga_main/fpga_main_data_mem_rv32i.sv

//===CORE 1&2 reset PC0,PC1,PC2,PC3====
W  ,  0x01C00100  ,  0x1
W  ,  0x01C00104  ,  0x1
W  ,  0x01C00108  ,  0x1
W  ,  0x01C0010C  ,  0x1
W  ,  0x02C00100  ,  0x1
W  ,  0x02C00104  ,  0x1
W  ,  0x02C00108  ,  0x1
W  ,  0x02C0010C  ,  0x1

//===CORE 1&2 release reset PC0,PC1,PC2,PC3====
W  ,  0x01C00100  ,  0x0
W  ,  0x01C00104  ,  0x0
W  ,  0x01C00108  ,  0x0
W  ,  0x01C0010C  ,  0x0
W  ,  0x02C00100  ,  0x0
W  ,  0x02C00104  ,  0x0
W  ,  0x02C00108  ,  0x0
W  ,  0x02C0010C  ,  0x0

//===CORE 1&2 Enalbe PC0,PC1,PC2,PC3====
W  ,  0x01C00150  ,  0x1
W  ,  0x01C00154  ,  0x1
W  ,  0x01C00158  ,  0x1
W  ,  0x01C0015C  ,  0x1
W  ,  0x02C00150  ,  0x1
W  ,  0x02C00154  ,  0x1
W  ,  0x02C00158  ,  0x1
W  ,  0x02C0015C  ,  0x1
```

ChrisShakkour, 2023-01-20:
@amichai-bd closing this task, i think the input pipe < using Linux features does a great job. do you agree?

amichai-bd, 2023-01-21:
Almost =)
You should write at least one sequence yourself to make sure you get the hang of it and you will be able to present it to others (and Goel).
I recommend a sequence that restarts any program that is already loaded.
1. Reset the "ALL_PC_RESET"
2. set all the PC enable one by one. (in our case, 8 writes)
3. Release the "ALL_PC_RESET"

The sequence should solve the problem of the Program not restarting without hard reset.
The programs that ended the sorting disabled their own PC - so just resetting the PC without enabling it wont work
[image]

Please make sure you create it, run it, and send a video - then you can close this issue.

Note:
This is a very small task that you probably need to do anyways before your live presentation

amichai-bd, 2023-01-21:
Here is the sequence you need:
1. reset the "CR_ALL_PC_RESET"       0x3C02040 (`1` will reset) //affects all cores
1. enable the CR_THREAD0_PC_EN  0x1C00150 (`1` will enable)//core 1
1. enable the CR_THREAD1_PC_EN  0x1C00154 (`1` will enable)//core 1
1. enable the CR_THREAD2_PC_EN  0x1C00158 (`1` will enable)//core 1
1. enable the CR_THREAD3_PC_EN  0x1C0015c (`1` will enable)//core 1
1. enable the CR_THREAD0_PC_EN  0x2C00150 (`1` will enable)//core 2
1. enable the CR_THREAD1_PC_EN  0x2C00154 (`1` will enable)//core 2
1. enable the CR_THREAD2_PC_EN  0x2C00158 (`1` will enable)//core 2
1. enable the CR_THREAD3_PC_EN  0x2C0015c (`1` will enable)//core 2
1. release the "CR_ALL_PC_RESET"   0x3C02040 (`0` will reset)//affects all cores

sequence/restart_prgram.txt
```
COM3
W
0x3C02040
0x1
W
0x1C00150
0x1
W
0x1C00154
0x1
W
0x1C00158
0x1
W
0x1C0015C
0x1
W
0x2C00150
0x1
W
0x2C00154
0x1
W
0x2C00158
0x1
W
0x2C0015C
0x1
W
0x3C02040
0x0
Q
Q
```

amichai-bd, 2023-01-21:
Please add an option to add "comments" to the sequence so that the uart_term.py won't reject as an error.
That will make it much easier to write/read sequences.
example:
```
COM3
// CR_ALL_PC_RESET = 1 - reset the program counters in the LOTR
W
0x3C02040
0x1
```