amichai-bd/riscv-multi-core-lotr #276: RC - Matrix Remember Order wrong EnShiftCol expression

asapir, 2023-06-11:
on source/rc/rtl/mro.sv line 54 contains the following code:
EnShiftCol[col_i][MRO_MSB:0] = validCol[col_i-1] && EnShiftCol[col_i-1][MRO_MSB:0] ;

This expression should enable shift if the column is valid, the problem is the '&&' is logical operator and not bitwise so LHS can result only 0 or 1, the shift vector cannot pass.

The correct expression should be:
EnShiftCol[col_i][MRO_MSB:0] = {MRO_SIZE{validCol[col_i-1]}} & EnShiftCol[col_i-1][MRO_MSB:0] ;

amichai-bd, 2023-06-11:
Great catch!
How did you find this issue? 

I'm surprised we never got a warning from the Quartus/Modelsim compilation and lint statick checks.

asapir, 2023-06-11:
I ran Synopsys VC SpyGlass Lint which reports on this, also if you look at rc - golden_alive_trk_RC_transactions.log you can see that there is missing RisgRspOut (at the end)

I wrote python script to properly check the RC trk file base on the protocol.

commit 3415b85c (2023-06-11): Bug fix:
close #277 - redundent responce when writing to DE10lite_mmio
close #276 - wrong mro age matrix calc - use use a bitwise for calculating the enable shift
closed 2023-06-11 by amichai-bd