amichai-bd/riscv-multi-core-lotr #277: LOTR - DE120Lite_MMIO wrong F2C_RspValid expression

asapir, 2023-06-11:
In source/lotr/rtl/DE10Lite_MMIO.sv line 204:
assign F2C_RspValidQ504H = CtrlCRMemRdEnQ504 || CtrlCRMemWrEnQ504 || CtrlVgaMemRdEnQ504 || CtrlVgaMemWrEnQ504;

This expression set the valid response signal, the issue is the WrEn should not trigger response valid and should be removed form RHS

assign F2C_RspValidQ504H = CtrlCRMemRdEnQ504 || CtrlVgaMemRdEnQ504;

amichai-bd, 2023-06-11:
Great catch!
This should be fixed as well!

Did you try and run the LOTR on an FPGA or simulation?
How did you find these errors?

asapir, 2023-06-11:
I ran simulation. I did some changes to rc and lotr loggers, making the rc log file looks the same (same format) in both TB. Then write an ugly python code to parse it and report protocol issues.

closed 2023-06-11 by amichai-bd

amichai-bd, 2023-06-11:
Thank you @asapir for your interest and contribution for this LOTR project!

I will note that this was a project I advised for over the 2-3 years and we got it up and working on FPGA!
But since the student that worked on it graduated, this is not getting much maintenance.

If you are interested in RISCV and would like to part of a really cool project that we are working on - please see the https://github.com/amichai-bd/fpga_mafia

Many things are similar, but we are trying to make the design more robust, less proprietary, more scalable and with a lot more features!

It is WIP, but in the next 2-3 months we should have a working FPGA!

asapir, 2023-06-12:
Hi Amichai,

This looks very interesting, I’ll take a look. There is something that you need?

Thanks, Asi

amichai-bd, 2023-06-12:
We have an endless list of features we would like to support.
Some need involvement in many disciplines to get to work. 
RTL design.
HW verification.
Post prosses automation checkers
RISCV SW stack, builder, linkers 

Example:
Write POC programs that can utilize the many-core architecture (with a good foundation and reusability of the SW architecture)
Support other IO on FPGA. (joystick, keyboard)
Support RGB mode on the VGA display. (lower resolution but add RGB) -> need to define the exact HW-SW handshake.
Improve the build, automation, github "actions" workflow
Add GUI application to interact with the environment. (run tests, tests status)
Add Gui application to interact with the UART & FPGA (Host->device read/write to FPGA MMIO, readout mems, debug etc.)

And many more!

You let me know what you think you want to assist with and we can set concrete task that w can integrate to the design.

Let me know!
You are welcome to email me (abendavid@nvidia.com / amichaibd@gmail.com) and we can even set a meeting if you'd like.

commit 3415b85c (2023-06-11): Bug fix:
close #277 - redundent responce when writing to DE10lite_mmio
close #276 - wrong mro age matrix calc - use use a bitwise for calculating the enable shift